cristian-mattarei/CoSA #35: Possible regression in latest CoSA release (0.2.1)

leonardt, 2018-09-19:
Here are the necessary files
[conf_sub.ets.txt](https://github.com/cristian-mattarei/CoSA/files/2398961/conf_sub.ets.txt)
[pe.json.txt](https://github.com/cristian-mattarei/CoSA/files/2398956/pe.json.txt)
[problem_pe_sub.txt](https://github.com/cristian-mattarei/CoSA/files/2398953/problem_pe_sub.txt)

Command
```
CoSA --problem problem_pe_sub.txt
```

Output
```
** Problem PE check sub configuration **
Description: "Check configuring to opcode=1 results in read_data=1"
Result: FALSE
Expected: TRUE
FALSE != TRUE <<<---------| ERROR
Counterexample:
---> INIT <---
  I: conf_done = 0_1
  I: self.I0 = 22_16
  I: self.I1 = 20_16
  I: self.O = 42_16
  I: self.clk = 0_1
  I: self.config_addr = 0_1
  I: self.config_data = 0_2
  I: self.config_en = 0_1
  I: self.read_data = 0_2
  I: self.reset = 0_1

---> STATE 1 <---
  S1: self.I0 = 18_16
  S1: self.I1 = 17_16
  S1: self.O = 35_16

---> STATE 2 <---
  S2: self.I0 = 15_16
  S2: self.I1 = 13_16
  S2: self.O = 28_16
  S2: self.reset = 1_1

---> STATE 3 <---
  S3: self.I0 = 11_16
  S3: self.I1 = 10_16
  S3: self.O = 21_16
  S3: self.reset = 0_1

---> STATE 4 <---
  S4: self.I0 = 8_16
  S4: self.I1 = 6_16
  S4: self.O = 14_16
  S4: self.clk = 1_1
  S4: self.config_data = 1_2
  S4: self.config_en = 1_1

---> STATE 5 <---
  S5: self.I0 = 4_16
  S5: self.I1 = 3_16
  S5: self.O = 7_16

---> STATE 6 <---
  S6: conf_done = 1_1
  S6: self.I0 = 1_16
  S6: self.I1 = 0_16
  S6: self.O = 1_16
  S6: self.clk = 0_1
  S6: self.config_data = 0_2
  S6: self.config_en = 0_1
```

This check used to pass on version 0.2.0. Let me know if this is actually an issue in the problem description rather than the checker.

cristian-mattarei, 2018-09-20:
Hi Lenny,
in version 0.2.0 there was a bug regarding the exit status of the software, so probably the property was failing even before. 
It seems that the tool is providing a correct result, and if you call the CoSA with --full-trace you can see that the counterexample shows that conf_done = 1_1 while self.read_data = 0_2, which contradicts the property (conf_done = 1_1) -> (self.read_data = 1_2). You can also call the tool with --vcd to produce a trace in vcd format.

cristian-mattarei, 2018-09-20:
I have also noticed that you manually assume the previous property. This feature is natively supported with the verification parameter `assume_if_true: True` that needs to be added to the `[GENERAL]` section.

cristian-mattarei, 2018-09-20:
I realized what was the issue: with the new version the clock behavior is not added anymore by default, so you need to add `add_clock: True`. Following is the problem file with the correct parameters:
```
[GENERAL]
model_file: pe.json,conf_sub.ets
add_clock: True
assume_if_true: True

[DEFAULT]
bmc_length: 40
verification: safety

[PE check sub configuration]
description: "Check configuring to opcode=1 results in read_data=1"
formula: (conf_done = 1_1) -> (self.read_data = 1_2)
prove: TRUE
expected: TRUE

[PE check sub functionality]
description: "Check configuring to opcode=1 corresponds to sub"
formula: (conf_done = 1_1) -> ((self.I0 - self.I1) = self.O)
prove: TRUE
expected: TRUE
```

closed 2018-09-20 by cristian-mattarei

leonardt, 2018-09-20:
Got it, thanks!