cristian-mattarei/CoSA #41: cosa distributed with older version of bit_vector

leonardt, 2018-10-11:
I was looking at https://pypi.org/project/CoSA/#files, and if I download `CoSA-0.2.4.tar.gz` and look in the archive, I see:
```
❯ ls
CoSA.egg-info PKG-INFO      README.rst    bit_vector    cosa          setup.cfg     setup.py
```

It looks like cosa is being distributed with an older version of the `bit_vector` package.

This breaks compatibility with repos like https://github.com/StanfordAHA/garnet, where we use both cosa and bit_vector, but are expecting a newer version of `bit_vector`. Any chance we could get cosa upgraded to a newer version?

Alternatively, can CoSA rename the older bit_vector it's packaged with to something else? E.g. cosa_bit_vector, this would at least avoid the name conflict. This is probably the quicker work around to unblock our issue.

cristian-mattarei, 2018-10-12:
it should be fixed with version 0.2.5. Can you confirm?

leonardt, 2018-10-12:
Still see it in https://files.pythonhosted.org/packages/b1/1c/538993c0def260396bce03b2833a0ccdf172923abba5eddae7a56d2ef898/CoSA-0.2.5.tar.gz

```
❯ wget https://files.pythonhosted.org/packages/b1/1c/538993c0def260396bce03b2833a0ccdf172923abba5eddae7a56d2ef898/CoSA-0.2.5.tar.gz
❯ tar -xf CoSA-0.2.5.tar.gz
❯ ls CoSA-0.2.5
CoSA.egg-info PKG-INFO      README.rst    bit_vector    cosa          setup.cfg     setup.py
```

Perhaps there's something being cached in your release process? Maybe in the `dist/` or `build/` directory?

leonardt, 2018-10-12:
Link from https://pypi.org/project/CoSA/0.2.5/#files

leonardt, 2018-10-17:
I just saw https://github.com/cristian-mattarei/CoSA/blob/master/setup.py#L43, is it possible that you have a `bit_vector` directory in the same directory where you have `setup.py`? It could be that the `find_packages` function is automatically discovering that folder and assuming it's part of the package you're trying to distribute.

cristian-mattarei, 2018-10-17:
I have uploaded a new version on PyPi. Now it should be fixed.

leonardt, 2018-10-17:
Works now, thanks so much! https://travis-ci.com/StanfordAHA/garnet/builds/88307649

closed 2018-10-17 by leonardt