cristian-mattarei/CoSA #54: Out of date with bit vector

rsetaluri, 2019-03-13:
Bit vector no longer has an `unsigned_value` member. It is being used here: https://github.com/cristian-mattarei/CoSA/blob/bbab8a3bc01d8d24782e18423842c0a37c24a406/cosa/encoders/coreir.py#L377

leonardt, 2019-03-13:
Is this a side effect of migrating to hwtypes? I think for integration we'll need CoSA to upgrade to the latest pycoreir which required hwtypes instead of the bit_vector package.

makaimann, 2019-03-13:
We’re still using the bit_vector package so that’s probably why. This might still be the only change needed though (besides changing the dependencies).

Looks like it should be as_uint(). Quick question: how are x values represented?

leonardt, 2019-03-13:
I don't think X values have been added to the new implementation of `BitVector` yet, so we'll have to add that.

@cdonovick I think we decided to defer this until it came up, did you have a plan for how to implement it?

cdonovick, 2019-03-13:
My plan was to implement a separate bitvector/bit types with four state logic. If they are necessary I will implement them.

makaimann, 2019-03-13:
Cool that makes sense. Eventually it would make sense to have X for CoSA, but I don’t *think* it’s really used at all currently. I’ll make this change and run the tests.

makaimann, 2019-03-13:
Fixed with 4d8c33bab36b3cbd6bb35419e31e60423c80221d

closed 2019-03-13 by makaimann