cristian-mattarei/CoSA #80: Enhancement Idea: Use Yosys through Python API

makaimann, 2019-05-18:
Yosys now has a Python API using `boost_python`. Enable it with:
`sudo make install ENABLE_PYOSYS=1 ENABLE_LIBYOSYS=1`

We can use this to encode the Verilog directly from RTLIL instead of going through the BTOR format. This would give us much better control over the design and even allow us to dynamically run passes based on the contents of the design.

4ong, 2020-09-08:
Is it possible to use it in pono (CoSa 2) already?

makaimann, 2020-09-08:
Hello, unfortunately this feature doesn't exist in Pono either. However, it would be a nice feature. It would probably make more sense to do this through C++ in Pono (but of course it could be accessed through the Python bindings also). I can add it as an enhancement idea there as well.