cristian-mattarei/CoSA #122: CoSA2?

ekiwi, 2019-11-13:
Is there any way to get hold of a copy of the [award-winning](http://fmv.jku.at/hwmcc19/) CoSA2?

I am currently working on a tool that emits bounded model checking problems in BTOR2 and SMT2 format. There are some problems where `yices` (+SMT2 encoding) does a lot better than `btormc` on the BTOR2 encoding.
It would be great if I had another fast model checker to compare with.

makaimann, 2019-11-13:
Hello, thanks for your interest in CoSA2! Unfortunately, we are not public yet, but we hope to be soon (we're shooting for a couple weeks, but no promises). 

In the meantime, there is a pre-release available:
```
wget https://web.stanford.edu/~makaim/files/cosa2-linux
chmod +x cosa2-linux
./cosa2-linux --help
```

The help message is currently pretty brief but should get you started. One thing to note is that it currently only checks one of the bads in the BTOR2 file. You can select which one using `-p`. 

This pre-release version is not the exact one that ran for the competition. It should be faster, but we haven't thoroughly tested that yet. I can also make the competition build available if there's interest in that.

Once we go public, I'll add a comment with a pointer and close this issue.

makaimann, 2020-02-12:
We actually went public a couple months ago, but I totally forgot to respond until @ahmed-irfan just reminded me.

Here's the link: https://github.com/upscale-project/cosa2/

closed 2020-02-12 by makaimann

ekiwi, 2020-02-12:
Congratulations on the public release!!