cristian-mattarei/CoSA #124: The problem about installing CoSA.

zymeng, 2020-02-29:
Hi Mattarei,

When I try to install CoSA according to your instructions, some errors about collecting corei are appeared.

The error reports are shown as follows:

Collecting coreir
  Downloading https://files.pythonhosted.org/packages/88/c5/299886ff6938f26cea7ffa64b7e2c9c607c4ca96018df3de2645531d4668/coreir-2.0.61.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/tmp/pip-install-fV7cqe/coreir/setup.py'"'"'; __file__='"'"'/private/tmp/pip-install-fV7cqe/coreir/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/tmp/pip-install-fV7cqe/coreir/pip-egg-info
         cwd: /private/tmp/pip-install-fV7cqe/coreir/
    Complete output (9 lines):
    [proxychains] DLL init: proxychains-ng 4.14
    [proxychains] DLL init: proxychains-ng 4.14
    [proxychains] DLL init: proxychains-ng 4.14
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/tmp/pip-install-fV7cqe/coreir/setup.py", line 92
        subprocess.check_call(["make", "-C", build_dir, f"-j{njobs}",
                                                                   ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

How can I solve the problem?
Thanks a lot.

leonardt, 2020-03-01:
Hmm looks like it's failing during the pycoreir installation process.  FYI we've only tested it with python 3.7, is there any way you can upgrade to a newer python version?

zymeng, 2020-03-01:
@leonardt Thanks a lot, I have installed CoSA successfully!

When I run the commands on the "Usage" section in your README.rst, there are some errors.
For example, When I run "CoSA -i examples/counters/counters.json --verification simulation -k 7",  the report is shown below.

/usr/local/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
Parsing file "examples/counters/counters.json"... ERROR: Cannot open file examples/counters/counters.json

ERROR: No top in file :examples/counters/counters.json

/Users/travis/build/leonardt/pycoreir/coreir-cpp/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/travis/build/leonardt/pycoreir/coreir-cpp/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/travis/build/leonardt/pycoreir/coreir-cpp/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/travis/build/leonardt/pycoreir/coreir-cpp/src/passes/transform/rungenerators.cpp:26 Done running generators
[1]    40888 segmentation fault  CoSA -i examples/counters/counters.json --verification simulation -k 7

How can I solve it? Why the file cannot be parsed?

makaimann, 2020-03-01:
I'm glad you were able to get CoSA installed! That's strange that the example is not working for you. I just tried it in a fresh environment and it worked fine for me. Is there any chance you're using an old version of CoreIR? I used the PyPi version obtained with: `python3.7 -m pip install coreir`.

zymeng, 2020-03-04:
Thanks a lot. That's really strange. Is the problem related to the operating system I use? Do I need to run CoSA in a Linux environment  rather than Mac OS.

makaimann, 2020-03-04:
It should work on Mac OS as well (people use CoSA with coreir on Mac) but it is probably better tested on Linux. Are you using the `pip` installed version of `coreir`?

zymeng, 2020-03-04:
I try to finish the installation from source, so the coreir is installed by the off-the-shelf script file in your project.

I have installed CoSA again on Linux, but when I run the example command "CoSA -i examples/counters/counters.json --verification simulation -k 7", the problem below is reported:

Parsing file "examples/counters/counters.json"... Traceback (most recent call last):
  File "/usr/local/bin/CoSA", line 11, in <module>
    load_entry_point('CoSA', 'console_scripts', 'CoSA')()
  File "/home/zymeng/CoSA_git/CoSA/cosa/shell.py", line 210, in main
    sys.exit(run_problems(problems_config))
  File "/home/zymeng/CoSA_git/CoSA/cosa/shell.py", line 166, in run_problems
    psol.solve_problems(problems_config)
  File "/home/zymeng/CoSA_git/CoSA/cosa/analyzers/dispatcher.py", line 368, in solve_problems
    modifier)
  File "/home/zymeng/CoSA_git/CoSA/cosa/analyzers/dispatcher.py", line 321, in parse_model
    (hts_a, inv_a, ltl_a) = parser.parse_file(filepath, general_config, flags)
  File "/home/zymeng/CoSA_git/CoSA/cosa/encoders/coreir.py", line 340, in parse_file
    self.__reset_structures()
  File "/home/zymeng/CoSA_git/CoSA/cosa/encoders/coreir.py", line 103, in __reset_structures
    self._init_mod_map()
  File "/home/zymeng/CoSA_git/CoSA/cosa/encoders/coreir.py", line 204, in _init_mod_map
    mod_map.append(("not",  (Modules.Not,  [self.IN, self.OUT])))
AttributeError: 'CoreIRParser' object has no attribute 'IN'

Is this a question of the coreir version？

makaimann, 2020-03-04:
Actually I'm not sure if that's the version or just that coreir is not being installed correctly for some reason. You should be able to run `python3 -c "import coreir; print(coreir.Context())"` without any problems. If that doesn't work, could you try installing coreir from source? https://github.com/rdaly525/coreir.git

zymeng, 2020-03-05:
Thanks @makaimann, I have tried to install coreir from source.
   -  Now I notice that when I run "CoSA -i examples/counters/counters.json --verification simulation -k 7", the segmentation fault is reported, which shows that ERROR: Cannot open file examples/counters/counters.json.
   -  Then I change the command and run "CoSA -i examples/counters/counters.v --verification simulation -k ",  it reports "RuntimeError: Top module not provided".
   -  Finally, I run "CoSA -i examples/counters/counters.sts --verification simulation -k", the checking results are generated.
What problems did I have when parsing the input file?

makaimann, 2020-03-05:
The json file is a CoreIR file so it still seems like CoreIR is not installed correctly. The second command you ran is a verilog file and it's complaining because you haven't told it which module is the top module for elaboration. You would run `CoSA -i examples/counters/counters.v --verification simulation -k <bound>`, and it requires that [Yosys](https://github.com/YosysHQ/yosys) is installed for processing the verilog. Finally, the last command works because it is an `sts` file which is CoSA's own format.

zymeng, 2020-03-05:
Great, I finally ran it correctly.
Thanks for your guidance!

closed 2020-03-06 by makaimann